uvic-auvic/Motor_Controller_1.0 #2: Change the PWM input channels

bates-andy123, 2017-01-12:
Configure the PWM input channels with the new input pins which are specified in the project schematic.

lymacasm, 2017-02-17:
I've created a branch lydens_issue2 for this issue. It isn't complete, but hardware can be tested on it. TIM15 and TIM1 PWM input works, I still haven't been successful with TIM2. Maybe check TIM2 out, I think something happens on the input for TIM2, but it's not correct. I wouldn't use TIM2 to validate hardware. But I believe the two functions are called read_duty_cycle and read_frequency

lymacasm, 2017-03-12:
I've got this working now, all three PWM input pins can read the correct duty cycle and frequency. They also will reset once their timers generate an update event for overflow. The code can be reviewed on lydens_issue2.

bates-andy123, 2017-03-13:
Thanks, It'll take me a while to testing where code. 

Could you make a change first. Unfortunately on the bad Motor Controller 1.1 Rev A I am not able to solder to pin A1 nicely. Do you think you could put it back on B3 temporarily?? Leaving the part for A1 there just commented out. That way we could test with Rev A and switch it over for Rev B.

Not urgent, as I think system integration testing will begin in about a week

lymacasm, 2017-03-13:
Sounds good, I've made the change to PB3, everything works. 

I've included a task in the commit (main.c) I was using for testing, just in case I need to make more changes. I'll remove it before we merge, unless you would like to use it for testing. Nothing fancy, I just read from all 3 timers, I usually put a breakpoint in, and then I can look at the values of the variables. But, in any case, it should be removed before any finalizations of the motor controller code. Let me know if you'd like me to remove it now, or whenever.

bates-andy123, 2017-03-14:
If you have tested it I feel pretty comfortable merging it into main. Feel free to create a  pull request and merge it when your ready

lymacasm, 2017-03-21:
Yes, I reverted the debugging stuff I had in there. It is ready to be merged.

bates-andy123, 2017-03-22:
Sweet, closing this issue

lymacasm, 2017-03-22:
Very sorry, just noticed that I never updated the headers. I'll create another pull request.

bates-andy123, 2017-03-22:
No Worries! Master currently uses read_frequency.... Do you want to re sync your branch to master before making this changes

lymacasm, 2017-03-22:
Yeah, I saw that. Sure, I'll update from master, change those, and include the changes in the pull request

lymacasm, 2017-03-22:
Alrighty, the branch should be ready to merge. Whenever you're ready.

closed 2017-03-22 by bates-andy123