uvic-auvic/Motor_Controller_1.0 #3: Set correct PWM Channels

bates-andy123, 2017-01-13:
The old circuit had different PWM output channels. We want to make sure that the new code only has the ability to pulse channels that are connected to the motor. The schematic below has the necessary connections. 

[Schematics.PDF](https://github.com/uvic-auvic/Motor_Controller/files/703691/Schematics.PDF)

lymacasm, 2017-01-30:
I've published a branch with my changes. The branch is named lydens_issue3.

I've tested the changes using a potentiometer. The pin PB1 no longer outputs PWM, while pins PA6, PA7 and PB0 do.

bates-andy123, 2017-01-30:
Hey, can you get rid of timer 16, and timer 2 completely. As we don't need those anymore so there's no point keeping them. From both the src and header

Also, could you make sure the other functions only change motors that still exist. Functions such as motor_init and motor_stop call set motors that don't exist. I don't think this causes an error but we should try to be clear

Also, can you explain you poteniometer test, how did you set it up?

lymacasm, 2017-01-31:
Sure, I'll make the changes.

Oops, I meant to write servo! I updated my comment to hide my mistake. I can still explain if you'd like

bates-andy123, 2017-01-31:
Too bad you didn't actually use a pot, that would been one hell of an explanation

lymacasm, 2017-01-31:
Haha, I'll try and think of a way to use a pot for next time!

lymacasm, 2017-02-01:
I've removed timers 2 and 16, as well as motor enumerations that refer to motors that we don't use, and calls to those motors. 

For reference, right_side corresponds to PA6, left_side corresponds to PA7, and top_right corresponds to PB0. We'll probably want to change those names to be more accurate, as I'm assuming the function (not C function) of each pin has changed?

bates-andy123, 2017-02-01:
Yes, I think we should change the names on those. If you want you could just change them to motor1, motor2, motor3 and we could figure it out later

lymacasm, 2017-02-01:
I've made the changes to motor1, motor2 and motor3. They map to the old enums as follows:
- motor1 -> top_right
- motor2 -> left_side
- motor3 -> right_side

lymacasm, 2017-02-01:
Hey, I just noticed when working on issue2 that timer16 is used in other areas of the code. Is this just a residual from the old system, or are we actually going to need timer16?

bates-andy123, 2017-02-01:
main.c is residual we can remove it

lymacasm, 2017-02-02:
It's also in PWM_in.c, used for the motor_pulse function

bates-andy123, 2017-02-02:
Leave that as your going to be modifying that in your next issue. We can resolve this issue and worry about it next issue

lymacasm, 2017-02-02:
Alright, so remove reference in main.c in this issue, and remove reference from PWM_in.c in the other issue?

bates-andy123, 2017-02-02:
Ya, it's arbitrary but let's do that for now.

lymacasm, 2017-02-03:
Hey Andy, is the DAQ IRQ handler for timer16 still in use?

bates-andy123, 2017-02-03:
No, it was used at one point to toggle the LED . But now that is done with RTOS so I see no use for it

bates-andy123, 2017-02-03:
Issue is now complete, further unnecessary code will be addressed in other issues

closed 2017-02-03 by bates-andy123