uvic-auvic/Motor_Controller_1.0 #7: Switch the buffer from being dynamic memory to static memory

bates-andy123, 2017-01-20:
We ran into new dynamic memory writing into running RTOS tasks. In preparation of putting RTOS on this system we must switch the buffer.c and buffer.h to the ones defined in uvic-auvic/common_firmware . 

1. Replace the old buffer code with new buffer code
2. Anywhere that interacted with the old buffer code has to be logically checked so that we know the new buffer code while behave identically with the system as the old buffer code did. 

PS. The old buffer code was infinite size, you'll notice that new one has a finite size. We are not really worried about overflow so setting this to 5 or higher should be sufficient.

lymacasm, 2017-01-23:
I've created a branch called lydens_issue7 with my changes.

bates-andy123, 2017-01-23:
I noticed that you use "unsigned short" in your code. I did some reading looks like it might be 16 bits?? It might be better to use "uint8_t" (which is unsigned 8 bits)? I believe the data type is in "stm32f0xx.h" .

lymacasm, 2017-01-23:
Yeah, unsigned short is 16 bits. The reason I chose to use 16 bits instead of 8 bits, is because the max for 8 bits is 255, whereas for 16 bits, it's 65535. I'm pretty sure my buffering would stop working properly if the head or tail overflowed, which is much more likely to happen for 8 bits (it would only take 255 commands vs 65535). I'm not sure how frequently commands are sent, and if we think 65535 would be a problem, we could either use a 32 bit int, or add some logic to stop it from happening. 

I first tried using uint16_t, but we don't have it defined. Do you think it would be possible for us to define it somewhere?

bates-andy123, 2017-01-24:
I will get back to you on the solving the types problem. There is a way to do it using the libraries we already have. I'll figure it out tonight

I have to admit I don't really like the idea of having a buffer which has a max useful size. Even if that size is really big. I would rather have extra code so that we have infinite size assuming it's used at a reasonable rate

lymacasm, 2017-01-24:
Alternatively, I could change to 8 bit, add a variable to keep track of the size, and apply a modulo operation every time I increment.

bates-andy123, 2017-01-24:
Much preferred, I don't think it needs to be overly fast. Also uses less ram 4 x 1 byte vs 3 x 2 byte

lymacasm, 2017-01-24:
Sounds good.

lymacasm, 2017-01-24:
I've made the changes, they've been committed to my branch

bates-andy123, 2017-01-25:
Is the current code in your branch the most up to date?

bates-andy123, 2017-01-25:
After talking it was confirmed most up to date version. Merged #9

closed 2017-01-25 by bates-andy123