uvic-auvic/Motor_Controller_1.0 #17: "Revolutions on All" Motor Command?

asinha94, 2017-03-10:
Would it be possible to have a command, similar to RVx in the [firmware doc](https://engrauvic.engr.uvic.ca/wiki/doku.php?id=elec:projects:motor_controller:fw_1.0#commands) which returns all the motor rpm's?

bates-andy123, 2017-03-10:
Yes, very easily. Do you just want them sequentially? It might be more then one byte for each command?

asinha94, 2017-03-10:
sure sequentially works

bates-andy123, 2017-03-10:
Not practically, Might involve some teaks I am lazy to make

bates-andy123, 2017-03-10:
I was just thinking it might be smart to put them in brackets or something. As we might change the length the command outputs. But let's worry about that later if it because a problem.

asinha94, 2017-03-10:
Yeah anything like `(RV1), (RV2), (RV3), (RV4), (RV5), (RV6)` works

bates-andy123, 2017-03-13:
This was added to the documentation and will be handled in issue #16

closed 2017-03-13 by bates-andy123