uvic-auvic/Motor_Controller_1.0 #18: Safety Shut-Off on Motor Command Timeout

lymacasm, 2017-03-13:
For safety reasons, the motors should "shut-off" (go into neutral) when we haven't received a motor speed command in a while. So a task needs to be implemented that checks when the last motor speed command was sent, and we should shut off all motors if a command hasn't been received after a certain amount of time. Currently, we're thinking 5 seconds is a good threshold.

lymacasm, 2017-03-22:
I've got working code for this, which can be located at lydens_issue18. 

I've implemented it using a software timer from the freertos library. Every time a command is received, a 5 second software timer is reset (starts counting from 5 seconds). If the 5 seconds expire, the timer callback function stops all motors. I've made my changes in motor.c, as that seemed natural as this is a feature for the motors. 

I've tested it using an LED in place of a motor (turn it on when a start a motor, turn it off if the timer expires).

bates-andy123, 2017-03-22:
Awesome!

As you are now implementing the commands I would be content merging this into master. I feel like you will notice if there is an error. 

Let me know if that works for you

lymacasm, 2017-03-22:
Yeah, my thoughts too. I was happy with the testing results. I'll make sure to check it again when I'm testing the commands on the actual motor controller, but as of now I feel comfortable with it. Merge away!

bates-andy123, 2017-03-22:
Sweet, it's done! 

Feel free to delete the branch. I don't want to delete for you just in case you need something :p

closed 2017-03-22 by bates-andy123

lymacasm, 2017-03-23:
Sure thing